NorthernWidget/Project-Margay #10: Add test for alarm pin in hardware test

bschulz1701, 2018-05-05:
Add a feature to test the hardware interrupt pin from the clock, do this by setting up an alarm on the clock, for 1 or 2 seconds, then wait and make sure interrupt gets triggered

bschulz1701, 2021-02-26:
This is a firmware issue and should be ported to the Margay_Library repo. This feature was added as of: 8afa026f7eff719172065aa3b60e5c7ad6d05fd9 (Merge branch 'master' into master, July 9th, 2018)

Recommend this issue be closed, can move to firmware repo for completeness

awickert, 2021-03-01:
Opened and closed an issue in the library repo here for completeness, per @bschulz1701:
https://github.com/NorthernWidget-Skunkworks/Margay_Library/issues/21
Now closing.

closed 2021-03-01 by awickert