NorthernWidget/Project-Margay #11: Add 2nd analog channel?

bschulz1701, 2018-06-13:
Does it make sense to add a second analog channel for support of some sensor varieties?

bschulz1701, 2021-02-26:
Based on current use cases, we feel a focus on a robust digital interface makes more sense. Can be revisited later, but on extended hiatus.

awickert, 2021-03-01:
I will add "wontfix" and close. In practical use cases, it seems that an inexpensive external ADC breakout (e.g., https://www.adafruit.com/product/1085) is perfectly fine. Indeed, this ADC is well supported by the Adafruit Arduino library, https://github.com/adafruit/Adafruit_ADS1X15. Agree with @bschulz1701 that we could revisit this, but that with a quality I2C interface, we can easily expand the Margay without cramming more on board.

closed 2021-03-01 by awickert